avatartwo/avatar2 #2: Missing parse_async_response method in QMPResponseListener

domenukk, 2017-06-21:
The `QMPResponseListener` seems to use the method `parse_async_response` which is only avaliable in `GDBResponseListener`.
https://github.com/avatartwo/avatar2/blob/master/avatar2/protocols/qmp.py#L58

mariusmue, 2017-06-22:
Yep, currently QMP can only handly synchronous messages, the asynchronous communication functionalities are currently not even activated:
https://github.com/avatartwo/avatar2/blob/master/avatar2/protocols/qmp.py#L84

Nevertheless, our boilerplate code from the asynchronous gdb-protocol parts made it in qmp-protocol somehow.
For now, we are missing a use case for asynchronous qmp messages. Nevertheless, we should either remove the asynchronous parts in qmp.py or properly implement them. Any thoughts?

mariusmue, 2017-06-26:
As there were no further objections, we removed the QMP-ResponseListener for now, to avoid confusion.
If we need to parse async-qmp-messages later on, we can just reimplement it.

closed 2017-06-26 by mariusmue