avatartwo/avatar2 #6: Avatar2 cannot handel XMM/YMM/128 bit registers

JeffJerseyCow, 2018-01-03:
gdb/mi -data-list-register-values has the potential to return at least three different types of results.

#### Normal
{'number': '183', 'value': '0x45a0'}

#### 128bit uint
{'number': '54', 'value': '{v4_float = {0x0, 0x0, 0x0, 0x0}, v2_double = {0x0, 0x0}, v16_int8 = {0x0 <repeats 16 times>}, v8_int16 = {0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0}, v4_int32 = {0x0, 0x0, 0x0, 0x0}, v2_int64 = {0x0, 0x0}, uint128 = 0x00000000000000000000000000000000}'}

#### 128bit int
{'number': '205', 'value': '{v8_float = {0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0}, v4_double = {0x0, 0x0, 0x0, 0x0}, v32_int8 = {0x2f <repeats 16 times>, 0x0 <repeats 16 times>}, v16_int16 = {0x2f2f, 0x2f2f, 0x2f2f, 0x2f2f, 0x2f2f, 0x2f2f, 0x2f2f, 0x2f2f, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0}, v8_int32 = {0x2f2f2f2f, 0x2f2f2f2f, 0x2f2f2f2f, 0x2f2f2f2f, 0x0, 0x0, 0x0, 0x0}, v4_int64 = {0x2f2f2f2f2f2f2f2f, 0x2f2f2f2f2f2f2f2f, 0x0, 0x0}, v2_int128 = {0x2f2f2f2f2f2f2f2f2f2f2f2f2f2f2f2f, 0x00000000000000000000000000000000}}'}

When converting these string values to int(val, 16) and exception is thrown and cannot be returned from **read_register_from_nr(...)** in protocols/gdb.py.

I've written a regex that extracts the value from in the two latter cases as a quick workaround.

### Patch
https://pastebin.com/hzgZ6KtW

This doesn't seem like a permanent solution especially as I'm ignoring half of the data in cases where a int128 number is utilised. It may be an idea to make a more generic way to read/write registers if supported by gdb.

I've only looked at this for the last few hours so I'm not sure.

JeffJerseyCow, 2018-01-03:
Never mind the patch, it just causes errors when setting the relevant register. 

{'type': 'result', 'message': 'error', 'payload': {'msg': 'Invalid cast.'}, 'token': 11, 'stream': 'stdout'}

or

{'type': 'result', 'message': 'error', 'payload': {'msg': 'Numeric constant too large.'}, 'token': 7, 'st
ream': 'stdout'}

mariusmue, 2018-01-04:
Thanks for the detailed report, I could easily reproduce the bug and are working on a fix.

In a nutshell, the strategy is to add the definition of "special-registers" in the architecture description, which get's checked by gdb when writing and reading registers. Based on that, just specific members of the register will be accessed instead of trying to get/set the full register.

mariusmue, 2018-01-05:
f4f00d4 inserts preliminary support for AVX and SSE instruction.
However, the registers which can return different values than the normal hexstring have to be explicitly defined.
Hence, it might be that relevant registers are not defined yet, could  you verify if this patch works for you?

JeffJerseyCow, 2018-01-07:
Issue is fixed, although there are additional extended registers --- zmm.

closed 2018-01-07 by JeffJerseyCow