avatartwo/avatar2 #8: Python 3

PanderMusubi, 2018-01-08:
Please, also support installing and running for Python 3.

mariusmue, 2018-01-08:
Installation and running of simple examples of avatar² works flawlessly for me with python3 - in fact, we take care in our development to be both py2 and py3 compliant.

Could you be more specific examples of what is failing for you?

PanderMusubi, 2018-01-08:
Ah, the problem was that I was missing installation of cmake. Perhaps add to the documentation some prerequisites, please.

Python 2: `sudo apt-get install python-pip python-setuptools python-dev cmake`

Python 3: `sudo apt-get install python-pip3 python3-setuptools python3-dev cmake`

Once that is added, this can be closed.

mariusmue, 2018-01-09:
Fixed with 8af98bfda3333d90c9e3c2674915eab935a81082

(Note: As different distributions are shipping a different default version of python, we just kept it general)

closed 2018-01-09 by mariusmue

PanderMusubi, 2018-01-09:
Thanks! Perhaps also mention that all also works with Python 3 packages. That is not evident but possible thanks to your work.