avatartwo/avatar2 #14: buid_qemu.sh fails on current ubuntu18

Trolldemorted, 2018-06-27:
```
[...]
  CC      util/aio-posix.o
  CC      util/compatfd.o
  CC      util/event_notifier-posix.o
  CC      util/mmap-alloc.o
  CC      util/oslib-posix.o
  CC      util/qemu-openpty.o
  CC      util/qemu-thread-posix.o
  CC      util/memfd.o
  CC      util/envlist.o
  CC      util/path.o
/home/benni/BA/avatar2/targets/src/avatar-qemu/util/memfd.c:40:12: error: static declaration of ‘memfd_create’ follows non-static declaration
 static int memfd_create(const char *name, unsigned int flags)
            ^~~~~~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/bits/mman-linux.h:115:0,
                 from /usr/include/x86_64-linux-gnu/bits/mman.h:45,
                 from /usr/include/x86_64-linux-gnu/sys/mman.h:41,
                 from /home/benni/BA/avatar2/targets/src/avatar-qemu/include/sysemu/os-posix.h:29,
                 from /home/benni/BA/avatar2/targets/src/avatar-qemu/include/qemu/osdep.h:104,
                 from /home/benni/BA/avatar2/targets/src/avatar-qemu/util/memfd.c:28:
/usr/include/x86_64-linux-gnu/bits/mman-shared.h:46:5: note: previous declaration of ‘memfd_create’ was here
 int memfd_create (const char *__name, unsigned int __flags) __THROW;
     ^~~~~~~~~~~~
/home/benni/BA/avatar2/targets/src/avatar-qemu/rules.mak:66: recipe for target 'util/memfd.o' failed
make: *** [util/memfd.o] Error 1
make: *** Auf noch nicht beendete Prozesse wird gewartet …
```

There is a [patch available](http://lists.openembedded.org/pipermail/openembedded-core/2018-January/146949.html), though.

mariusmue, 2018-06-27:
Thanks for reporting this issue - it is already known to us internally. We want to use this as opportunity to rebase avatar-qemu on upstream qemu.
However, some of the qemu-internals changed, and we want to re-factor our configurable machine as part of this process. The planned time of completion for this is approximately one month from now.

The recommended workaround is using ubuntu 17.10 (or earlier) for now.

However, if you feel like providing a PR to avatar-qemu just with this patch, I'm happy to merge it over.

mariusmue, 2019-01-23:
We updated Qemu to version 3.1 (d3abaeab297194c8700e9aa10b171eb54c756304). Technically, this should include an according fix, please feel free to reopen this issue in case this is not the case.

closed 2019-01-23 by mariusmue