avatartwo/avatar2 #17: QemuTarget: State in watchman not immediately updated after cont()

cwerling, 2018-07-02:
When a synchronous watchman hook is called after a `qemu.cont`, the QemuTarget's state is still `TargetStates.STOPPED`.

Sample script output:
```
$ python3 avatar_state_test.py
State in watchman before memory access: TargetStates.STOPPED
State in watchman after memory access: TargetStates.STOPPED
State in watchman before memory access: TargetStates.STOPPED
State in watchman after memory access: TargetStates.STOPPED
State in watchman before memory access: TargetStates.STOPPED
State in watchman after memory access: TargetStates.STOPPED
State in watchman before memory access: TargetStates.STOPPED
State in watchman after memory access: TargetStates.STOPPED
State in watchman before memory access: TargetStates.STOPPED
State in watchman after memory access: TargetStates.RUNNING
State in main thread after cont(): TargetStates.RUNNING
State in watchman before memory access: TargetStates.RUNNING
State in watchman after memory access: TargetStates.RUNNING
State in watchman before memory access: TargetStates.RUNNING
State in watchman after memory access: TargetStates.RUNNING
State in watchman before memory access: TargetStates.RUNNING
State in watchman after memory access: TargetStates.RUNNING
[...]
```

Sample script:
```python
from avatar2 import *
from avatar2.peripherals import AvatarPeripheral

import os

class CustomMemoryPeripheral(AvatarPeripheral):

    def hw_read(self, offset, size):
        ret = self.hello_world[:size]
        self.hello_world = self.hello_world[size:] + self.hello_world[:size]
        return int.from_bytes(ret, 'little')

    def nop_write(self, offset, size, value):
        return True

    def __init__(self, name, address, size, **kwargs):
        AvatarPeripheral.__init__(self, name, address, size)

        self.hello_world = 'Hello World'

        self.read_handler[0:size] = self.hw_read
        self.write_handler[0:size] = self.nop_write

def before_remote_memory_access(avatar, remote_memory_msg, **kwargs):
    print('State in watchman before memory access: %s' % qemu.state)

def after_remote_memory_access(avatar, remote_memory_msg, **kwargs):
    print('State in watchman after memory access: %s' % qemu.state)

def main():
    global qemu

    dir_path = os.path.dirname(os.path.realpath(__file__))

    avatar = Avatar()

    # Text section
    avatar.add_memory_range(0x00000000, 0x00013000, file=dir_path+'/sample.bin')

    # Watchmen
    avatar.watchmen.add_watchman('RemoteMemoryRead', 'before', before_remote_memory_access, is_async=False)
    avatar.watchmen.add_watchman('RemoteMemoryWrite', 'before', before_remote_memory_access, is_async=False)

    avatar.watchmen.add_watchman('RemoteMemoryRead', 'after', after_remote_memory_access, is_async=False)
    avatar.watchmen.add_watchman('RemoteMemoryWrite', 'after', after_remote_memory_access, is_async=False)

    # Memory Peripheral
    avatar.add_memory_range(0x00013000, 0xfffecfff, name='unmapped_memory', emulate=CustomMemoryPeripheral, permission='rw-')

    # QEMU CONFIG
    qemu = avatar.add_target(QemuTarget, name='qemu1')
    qemu.entry_address = 0x100
    qemu.cpu_model = 'cortex-a8'

    qemu.init()
    qemu.cont()
    print('State in main thread after cont(): %s' % qemu.state)
    qemu.wait()
    qemu.shutdown()

if __name__ == '__main__':
    main()
```

mariusmue, 2018-09-28:
Hmm, I can't reproduce this on my machine. Would you mind sharing the sample.bin?

mariusmue, 2020-07-14:
We had quite some restructuring in the watchmen and state-update mechanisms since this issue was opened (lastly with #55).
I assume that these changes fix the behavior described here. However, as the sample.bin for reproduction was never provided, I cannot verify this.

For now, I'm going to close this issue, should the problem still exist, please reopen.

closed 2020-07-14 by mariusmue