avatartwo/avatar2 #19: Unable to create custom peripheral at address 0

RobertBuhren, 2018-09-21:
Any attempt to create a custom memory mapping such as:
``avatar.add_memory_range(0x0, 0x100, name='unmapped_memory', emulate=CustomMemoryPeripheral)``
Will result in a timeout in when avatar tries to connect via GDB to a Qemu target. The same mapping works 
if the start address is not 0. E.g.:
``avatar.add_memory_range(0x8, 0x100, name='unmapped_memory', emulate=CustomMemoryPeripheral)``

Regards,
Robert

mariusmue, 2018-09-21:
Interesting, I'll look into it.

mariusmue, 2018-09-28:
Okay, after some investigation, this bug is related to gdb's frame-pointer handling.
GDB automagically tries to create the stack frames upon connection (or hitting of an breakpoint).
For doing so, it dereferences r11 on those occasions for ARM - unfortunately, this only works for APCS (the obsolete ARM ABI), as in AAPCS r11 is a generic data register.

As a result, r11 is 0 upon initialization, but qemu gets forced to read from 0 by gdb, which times out as RMP is not initialized yet. The clean solution would be to fix this in gdb itself, however, according requests for patches have been disregarded in the past (c.f. https://sourceware.org/ml/gdb/2004-06/msg00048.html).
However, as we constantly running in problems with this, we may reach out to the gdb-devs.

As of know, the suggested hotfix is to build gdb from source, and patch the function `arm_scan_prologue` in `gdb/arm-tdep.c` to return before the frame info is "retrieved".