avatartwo/avatar2 #20: HelloWorld  peripheral does not work

RobertBuhren, 2018-10-17:
Hi,
I have created a simple test case to test the custom peripheral feature of avatar:
Essentially I want to get a notification each time the emulated code accesses a certain memory area. Here's a simple example:

```
ldr r0,=0x40004c00
ldr r1,[r0]
str r1,[r0]
```
I use this avatar2 script (essentially I just added prints to the HelloWorld example):
 
```
from avatar2 import *
from avatar2.peripherals import AvatarPeripheral
from capstone import CS_MODE_THUMB

import os, struct
from time import sleep

from IPython import embed

class HelloWorldPeripheral(AvatarPeripheral):

    def hw_read(self, offset, size):         
        ret = self.hello_world[:size]
        self.hello_world = self.hello_world[size:] + self.hello_world[:size]
        print("READ")
        return ret

    def nop_write(self, offset, size, value):
        print("WRITE")
        return True    

    def __init__(self, name, address, size, **kwargs):
        AvatarPeripheral.__init__(self, name, address, size)
        
        self.hello_world='Hello World'
        
        self.read_handler[0:size] = self.hw_read 
        self.write_handler[0:size] = self.nop_write
        

avatar = Avatar(output_directory='/tmp/avatar123')
avatar.log.setLevel('DEBUG')

hw = avatar.add_memory_range(0x40004c00, 0x100, name='hello_world',
                             emulate=HelloWorldPeripheral, permissions='rw-')        

avatar.add_memory_range(0x1000, 0x1000, file='./test.raw')

qemu = avatar.add_target(QemuTarget, name='qemu1',entry_address=0x1000)
qemu.cpu_model = 'cortex-a8'

qemu.init()
qemu.cont()
qemu.wait()
```
The result I'd expect that both "READ" and "WRITE" is printed, as the code emulated both reads and writes to the emulated memory area. However, the python script simply hangs and must be killed.

I have attached the logs as well as the sources for my example.

Regards,

Robert

RobertBuhren, 2018-10-18:
Hm, I adapted my example accordingly (see files attached). I was not able to get it to work though.
```
from avatar2 import *
from avatar2.peripherals import AvatarPeripheral
from capstone import CS_MODE_THUMB

import os, struct
from time import sleep

from IPython import embed

class HelloWorldPeripheral(AvatarPeripheral):

    def hw_read(self, offset, size):         
        ret = self.hello_world[:size]
        self.hello_world = self.hello_world[size:] + self.hello_world[:size]
        # Convert the return value to an integer (py2/py3-compatible)
        # Python >3.2 could just call int.from_bytes(ret, byteorder='little')
        s2fmt = {1: 'B', 2: 'H', 4: 'I', 8: 'Q'}
        ret = struct.unpack('<' + s2fmt[size], ret)[0]

        print("READ")
        return ret

    def nop_write(self, offset, size, value):
        print("WRITE")
        return True    

    def __init__(self, name, address, size, **kwargs):
        print("Init")
        AvatarPeripheral.__init__(self, name, address, size)
        
        self.hello_world='Hello World'
        
        self.read_handler[0:size] = self.hw_read 
        self.write_handler[0:size] = self.nop_write
        

avatar = Avatar(output_directory='/tmp/avatar123')
avatar.log.setLevel('DEBUG')

hw = avatar.add_memory_range(0x40004c00, 0x100, name='hello_world',
                             emulate=HelloWorldPeripheral, permissions='rw-')        

avatar.add_memory_range(0x1000, 0x1000, file='./test.raw')

qemu = avatar.add_target(QemuTarget, name='qemu1',entry_address=0x1000)
qemu.cpu_model = 'cortex-a8'

qemu.init()
print("QEMU.init()")
qemu.set_breakpoint(0x1008)
print("QEMU.set_breakpoint()")
qemu.cont()
print("QEMU.cont()")
qemu.wait()
print("QEMU.wait()")
embed()

```

[logs.tar.gz](https://github.com/avatartwo/avatar2/files/2492203/logs.tar.gz)

RobertBuhren, 2018-10-18:
I added this patch to qemu in order to be able to build it with glibc 2.27

https://lists.gnu.org/archive/html/qemu-devel/2017-11/msg05016.html

That's all I changed

mariusmue, 2018-10-18:
Hmm, I really think that it is related to the missing rx_queue. 9b79e8acc2144cdbb28d002c0e23a1b0b8eecf00 changes the exception handling of the according path - would you mind pulling avatar2 and re-executing?

mariusmue, 2018-10-18:
Update, there is still a problem in your script/the example hw peripheral. The hello world string needs to be a bytes-object for having unpack to work in py3, hence:
`        self.hello_world='Hello World'` ->  `        self.hello_world=b'Hello World`
 (af08154a00a5c58d17886b950602e4059ce4e780)

RobertBuhren, 2018-10-18:
ahhh. Ok now it works. Converting the string to a byte object did the trick. 
However, I don't see any exception in case it is a string. Maybe crashing early here would help the user?

Thanks again.

Regards,
Robert

mariusmue, 2018-10-19:
That's a very fair point. While not directly forcing a crash, f39943c4b6bc2db15090f9f51eb3317fb80c149b introduces actual useful logging of according exceptions in the log file.
(Rationale is that in some cases, the user may want to catch failed remote memory requests manually

In any case, thanks for reporting the issue!

closed 2018-10-19 by mariusmue