avatartwo/avatar2 #21: Cannot build avatar-panda

dukebarman, 2018-12-10:
`build_qemu.sh` works fine. But build_panda.sh has an error:

```
fatal: reference is not a tree: bff62d20a88e700a6e5ec54afb457bd433e1d4ab
Unable to checkout 'bff62d20a88e700a6e5ec54afb457bd433e1d4ab' in submodule path 'avatar-panda'
error: pathspec 'dtc' did not match any file(s) known to git.
./build_panda.sh: line 27: ../../../src/avatar-panda/configure: No such file or directory
make: *** No targets specified and no makefile found.  Stop.
```

OS:
```
uname -a
Linux ubuntu 4.15.0-42-generic #45~16.04.1-Ubuntu SMP Mon Nov 19 13:02:27 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux```

Avatar2 from git

mariusmue, 2018-12-11:
Thanks for letting us know! The problem was a not updated reference to the avatar-panda repository.
Fixed with 7348f7c

dukebarman, 2018-12-12:
Another error `~/avatar2/targets/src/avatar-panda/panda/plugins/network/network.cpp:22:30: fatal error: wireshark/config.h: No such file or directory`

After install additional packages `sudo apt install libwiretap-dev libwireshark-dev` (from install_ubuntu.sh script):

```
~/avatar2/targets/src/avatar-panda/panda/plugins/network/network.cpp: In function ‘int handle_packet(CPUState*, uint8_t*, int, uint8_t, uint64_t)’:
~/avatar2/targets/src/avatar-panda/panda/plugins/network/network.cpp:102:24: error: aggregate ‘handle_packet(CPUState*, uint8_t*, int, uint8_t, uint64_t)::wtap_pkthdr header’ has incomplete type and cannot be defined
     struct wtap_pkthdr header;
                        ^
~/avatar2/targets/src/avatar-panda/panda/plugins/network/network.cpp:107:27: error: ‘wtap_phdr_init’ was not declared in this scope
     wtap_phdr_init(&header);
                           ^
~/avatar2/targets/src/avatar-panda/panda/plugins/network/network.cpp:134:30: error: ‘wtap_phdr_cleanup’ was not declared in this scope
     wtap_phdr_cleanup(&header);
```

mariusmue, 2018-12-12:
This is a problem with PANDA's dependencies themselves. It requires an older version of wireshark. While upstream-panda has this fixed, avatar-panda is not up to date now. 
We are likely to synchronize Panda and QEmu in the near future - for now removing the `network` plugin in avatar-panda/plugins/config.panda (or downgrading wireshark) is an acceptable hotfix. I'm reopening this issue for now.

dukebarman, 2018-12-13:
A small fix `avatar-panda/plugins/config.panda` -> `avatar-panda/panda/plugins/config.panda`. It works, thx. Waiting for fix!

mariusmue, 2019-01-24:
I finally found some time to rebase avatar-panda on the current version of upstream panda.
Hence, the compilation problem should be fixed now with commit 0779fd39a831d01e47448aec2df595cbc63302d8.

Cheers, and sorry for the delay!

closed 2019-01-24 by mariusmue