avatartwo/avatar2 #22: Failed with keystone errors

dukebarman, 2019-01-24:
Try to execute examples from `https://github.com/avatartwo/avatar2-examples/tree/master/qemu_uboot`, but have a error:

```
$ python qemu_uboot_avatar2.py 
Traceback (most recent call last):
  File "qemu_uboot_avatar2.py", line 1, in <module>
    from avatar2 import *
  File "/home/user/.local/lib/python2.7/site-packages/avatar2/__init__.py", line 6, in <module>
    from .message import *
  File "/home/user/.local/lib/python2.7/site-packages/avatar2/message.py", line 45, in <module>
    from .targets.target import TargetStates
  File "/home/user/.local/lib/python2.7/site-packages/avatar2/targets/__init__.py", line 3, in <module>
    from .openocd_target import *
  File "/home/user/.local/lib/python2.7/site-packages/avatar2/targets/openocd_target.py", line 10, in <module>
    from avatar2.protocols.gdb import GDBProtocol
  File "/home/user/.local/lib/python2.7/site-packages/avatar2/protocols/gdb.py", line 17, in <module>
    from avatar2.archs.arm import ARM
  File "/home/user/.local/lib/python2.7/site-packages/avatar2/archs/__init__.py", line 2, in <module>
    from .arm import *
  File "/home/user/.local/lib/python2.7/site-packages/avatar2/archs/arm.py", line 12, in <module>
    class ARM(Architecture):
  File "/home/user/.local/lib/python2.7/site-packages/avatar2/archs/arm.py", line 38, in ARM
    keystone_arch = KS_ARCH_ARM
NameError: name 'KS_ARCH_ARM' is not defined
```

`python -m avatar2.installer` doesn't work too

avatar: `pip install avatar2`
OS: Ubuntu 18

mariusmue, 2019-01-24:
Thanks for reporting! However, I can't reproduce the (exact) error.
Which Ubuntu 18 are you using? (Please note that our testing distribution is still Ubuntu 16.04.)

I just tried on a clean installation of Ubuntu 18.04.01 LTS (vagrant VM bento/ubuntu-18.04).
Do you have the log for `pip install avatar2`? My assumption is, that the installation of keystone failed, most likely due to a missing dependency, such as `cmake` (which is listed in README.md, but unfortunately not delivered via pip). 

However, even then the following error occurs:
```
Traceback (most recent call last):
  File "qemu_uboot_avatar2.py", line 1, in <module>
    from avatar2 import *
  File "/home/vagrant/.local/lib/python2.7/site-packages/avatar2/__init__.py", line 6, in <module>
    from .message import *
  File "/home/vagrant/.local/lib/python2.7/site-packages/avatar2/message.py", line 45, in <module>
    from .targets.target import TargetStates
  File "/home/vagrant/.local/lib/python2.7/site-packages/avatar2/targets/__init__.py", line 3, in <module>
    from .openocd_target import *
  File "/home/vagrant/.local/lib/python2.7/site-packages/avatar2/targets/openocd_target.py", line 10, in <module>
    from avatar2.protocols.gdb import GDBProtocol
  File "/home/vagrant/.local/lib/python2.7/site-packages/avatar2/protocols/gdb.py", line 17, in <module>
    from avatar2.archs.arm import ARM
  File "/home/vagrant/.local/lib/python2.7/site-packages/avatar2/archs/__init__.py", line 2, in <module>
    from .arm import *
  File "/home/vagrant/.local/lib/python2.7/site-packages/avatar2/archs/arm.py", line 4, in <module>
    from keystone import *
  File "/home/vagrant/.local/lib/python2.7/site-packages/keystone/__init__.py", line 4, in <module>
    from .keystone import Ks, ks_version, ks_arch_supported, version_bind, debug, KsError, __version__
  File "/home/vagrant/.local/lib/python2.7/site-packages/keystone/keystone.py", line 75, in <module>
    raise ImportError("ERROR: fail to load the dynamic library.")
ImportError: ERROR: fail to load the dynamic library.
```
This build problem is a known error of keystone, c.f. https://github.com/keystone-engine/keystone/issues/386. After manually fixing up the path of libkeystone.so, at least executing the installer itself is working. However, installing qemu/gdb via it seems not to be possible on a clean system.
(Missing dependencies, wrong package names for 18.04).

All in all, I see the following tasks necessary for a better build-experience:
- [ ]  Enhance setup.py to fixup keystone path in PostInstall, if neccessary.
- [ ]  Let the avatar2-installer check OS/Version it's running on and adjust package names accordingly.
- [ ] Allow the installer to figure out missing dependencies and prompt the user for installation of those.

Help/PRs are more than welcome here.

dukebarman, 2019-01-24:
Yep, I installed `cmake` too. And sometimes I have this error too

Vge0rge, 2019-02-08:
I faced the same error and I managed to solve it even though my solution is not pretty. So in my case with Arch linux I found out that keystone was installed in two folders: 
~/.local/lib/python3.7/site-packages/keystone/
/usr/lib/python3.7/site-packages/keystone_engine-0.9.1.post3-py3.7.egg/keystone/

These two folders seemed to have the same files ( the exact same filenames ) but in the case of the first folder all the files were empty. So the solution for me was to copy all the files from the  /usr/lib/... folder  into the local one. That solved it for me, hopefully it will solve it for you too!

mariusmue, 2019-04-09:
I added your patch with  959a568075efb7706d3e80e34470dbb807cdf29d, thanks for submitting! However, I think the general keystone error will persist until [keystone-engine on the python packet](https://pypi.org/search/?q=keystone-engine) index is updated.
While this is out of our hands, there seems a bit of progress happening: https://github.com/keystone-engine/keystone/issues/411

mariusmue, 2020-07-14:
The latest version of keystone on [PyPi](https://pypi.org/project/keystone-engine/0.9.2/) (v0.9.2) seems to fix the keystone-related installation errors.
Hence, I'm closing this issue for now - if there still persist problem, please add a comment here.

closed 2020-07-14 by mariusmue