avatartwo/avatar2 #24: some error for python -m avatar2.installer

ZakiG, 2019-02-07:
I want to use `python -m avatar2.installer` to build avatar-panda,but I got those error(Ubuntu 14.04 64bit):
`
Cloning into '/home/XXXX/.avatar2/avatar-panda'...
remote: Enumerating objects: 1938, done.
remote: Counting objects: 100% (1938/1938), done.
remote: Compressing objects: 100% (659/659), done.
remote: Total 311315 (delta 1355), reused 1707 (delta 1262), pack-reused 309377
Receiving objects: 100% (311315/311315), 129.64 MiB | 3.72 MiB/s, done.
Resolving deltas: 100% (248868/248868), done.
Checking connectivity... done.
Submodule 'dtc' (git://git.qemu-project.org/dtc.git) registered for path 'dtc'
Cloning into 'dtc'...
remote: Counting objects: 4807, done.
remote: Compressing objects: 100% (2194/2194), done.
remote: Total 4807 (delta 3561), reused 3434 (delta 2571)
Receiving objects: 100% (4807/4807), 1.04 MiB | 241.00 KiB/s, done.
Resolving deltas: 100% (3561/3561), done.
Checking connectivity... done.
Submodule path 'dtc': checked out 'ec02b34c05be04f249ffaaca4b666f5246877dea'

ERROR: sizeof(size_t) doesn't match GLIB_SIZEOF_SIZE_T.
       You probably need to set PKG_CONFIG_LIBDIR
       to point to the right pkg-config files for your
       build target
`
I check the configure  file of avatar-panda and find the check:
`
if ! compile_prog "$CFLAGS" "$LIBS" ; then
    error_exit "sizeof(size_t) doesn't match GLIB_SIZEOF_SIZE_T."\
               "You probably need to set PKG_CONFIG_LIBDIR"\
	       "to point to the right pkg-config files for your"\
	       "build target"
fi
`
Howerver, I don not know how to solve it :( .
Thanks for your help.

mariusmue, 2019-02-07:
Hi,
This is a problem of qemu, the underlying emulator, and I don't know how to fix this without further investigation either.
Unfortunately, we are testing the build-scripts on Ubuntu 16.04, where it is known to work. Is it possible for you to retry the build with Ubuntu 16.04, probably inside an VM?

Best,
Marius

ZakiG, 2019-02-08:
Emmm, I try to build in Ubuntu 16.04. 
After the experiment, use `python -m avatar.installer`,  there is no problem if you choose build `avatar-qemu`. However, I got those problem:
`if ! compile_prog "$CFLAGS" "$LIBS" ; then error_exit "sizeof(size_t) doesn't match GLIB_SIZEOF_SIZE_T."\ "You probably need to set PKG_CONFIG_LIBDIR"\ "to point to the right pkg-config files for your"\ "build target" fi` 
when I choose build `avatar-panda`.  :(
But if I want to do the experiment(`ndss18_wycinwyc`) , I have to build `avatar-panda`. 
It is worth nothing that I got the same error when build `avatar-panda` in Ubuntu 14.04 , 16.04 , 18.04...
Thanks for help
Best, 
Zaki

mariusmue, 2019-02-08:
Thanks for reporting. I will look into it, but are a little bit busy right now. As a workaround, you could use the pre-built Vagrantbox as workaround:
`$ vagrant init avatar2/ndss18_wycinwyc`

I'll leave this issue open until I reproduced and fixed the bug.

Vge0rge, 2019-02-08:
I faced the same problem with panda with Arch.  The config.log had a very long list of errors that I didn't have the time to debug. With a quick look it seems like a glibc architecture problem. I will start from scratch this weekend and I will let you know if I have any progress ;)

mariusmue, 2019-02-08:
Thanks a lot! Alternatively, I will look into it by myself _next_ weekend, sorry for the delay.

Vge0rge, 2019-02-08:
If I manage to do that should I upload a VM with the complete installation here? Will this be useful?

mariusmue, 2019-02-08:
The vagrant box specified further up in the issue, or `avatar2/bar18_avatar2` should have working, but outdated installations.
In that sense, having somewhere a "current" vm does for sure not hurt. If it is a vagrant/virtualbox image which I can test, I could also add it here: https://app.vagrantup.com/avatar2

Vge0rge, 2019-02-08:
Awesome! Ok, I will give any updates soon ;)

Vge0rge, 2019-02-09:
Just finished the compilation, most probably you are missing some 32bit libraries such as glib. This was the case with me, you need to check your config.log file and see what it says. You also need to check that you have all the packages that exist in the file panda/scripts/install_ubuntu.sh ( or similar files ) because most probably you are missing some header files too.  Lastly there are some warnings that you must ignore during compilation  so you need to run configure with the "-Wno-error" flag. The command that I used is the following: 
$ CFLAGS="-Wno-error" ./configure --python=python2.7 --disable-sdl

Having said that I need to note that I didn't test the functionality of panda at this point, I just managed to compile it successfully. 

Hope that helps, 
George

mariusmue, 2019-02-11:
Thanks for dropping your notes! I will update our build-scripts on the next occasion and maybe add a small FAQ in the readme.md which also contains the solutions to the keystone issues as reported in #23/#22.

mariusmue, 2020-07-14:
The latest version of keystone on [PyPi](https://pypi.org/project/keystone-engine/0.9.2/) (v0.9.2) seems to fix the keystone-related installation errors.
Hence, I'm closing this issue for now - if there still persist problem, please add a comment here.

mariusmue, 2020-07-14:
Oups, this issue is not only keystone-related -> reopening.