avatartwo/avatar2 #27: Issue setting breakpoints

ChrisMcMStone, 2019-03-14:
I'm having difficulty setting breakpoints over a remote GDB interaction. Specifically when I try to break on a hex address.

Here is my avatar python code. 
```
gdb = ava.add_target(avatar2.GDBTarget, gdb_port=GDB_SERVER_PORT)
ava.init_targets()
gdb.bp(0x92b)
gdb.cont()
```
Debug log when attempting to set the breakpoint. 
```
avatar.targets.GDBTarget0.GDBProtocol.DEBUG | gdb.py line:299 | Sending request: 4-break-insert *0x92b
avatar.targets.GDBTarget0.GDBProtocol.DEBUG | gdb.py line:504 | Attempted to set breakpoint. Received response: {'type': 'result', 'message': 'done', 'payload': {'bkpt': {'number':        '1', 'type': 'breakpoint', 'disp': 'keep', 'enabled': 'y', 'addr': '0x000000000000092b', 'thread-groups': ['i1'], 'times': '0', 'original-location': '*0x92b'}}, 'token': 4, 'stream':      'stdout'}
```

And GDBserver log. 
```
Failed to read shadow memory of breakpoint at 0x92b (Input/output error).
Failed to insert breakpoint at 0x92b (-1).
```

Breakpoint functionality does however work when trying with a function name instead of adress. I've tried on multiple binaries and am sure I have the correct addresses. Can anyone try replicating the issue?

mariusmue, 2019-03-14:
Replicating this without binary or more information is difficult.
For me, this looks like a classic case where the memory you want to break is not accessable to gdb yet, which can have a variety of reasons. Over here, 'hex-breakpoints' never posed any problems.

If it is an embedded system you are connecting to, you may want to pass `hardware=true` to `gdb.set_breakpoint`/`gdb.bp`

ChrisMcMStone, 2019-03-14:
@mariusmue Thanks marius. It turns out it was because the binary was a PIE (position independant exectuable), meaning the base address is determined at run time and not 0x00. As a result my `0x092b` address was wrong so the breakpoint failed. 
To resolve, either compile your binary without PIE, the gcc flag is `-no-pie`. Or use GDB to get the base address after executing the binary, e.g with `info proc mappings`.

closed 2020-07-14 by mariusmue