avatartwo/avatar2 #29: Building avatar-qemu: unused function

phros, 2019-04-12:
Hi there,

I'm trying to build `avatar-qemu`. It fails with the following error:
```
~/.avatar2/avatar-qemu/hw/avatar/configurable_machine.c:67:20: error: unused
      function 'unset_feature' [-Werror,-Wunused-function]
static inline void unset_feature(CPUARMState *env, int feature)
                   ^
1 error generated.
/home/roskosch/.avatar2/avatar-qemu/rules.mak:69: recipe for target 'hw/avatar/configurable_machine.o' failed
make[1]: *** [hw/avatar/configurable_machine.o] Error 1
Makefile:483: recipe for target 'subdir-arm-softmmu' failed
make: *** [subdir-arm-softmmu] Error 2
```
lsb_release -a:
```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic
```

Is there anything i can do about it? Let me know when you need further information.

Thank!

aurelf, 2019-04-12:
Hi phros, 
looks like you have " -Werror" and "-Wunused-function" in your CFLAGS, I don't think QEMU would build with this. Can you try to unset your CFLAGS before compiling?
"unset CFLAGS" should do.

phros, 2019-04-12:
`unser CFLAGS` does not change anything. It happens only when using the `python -m avatar2.installer` command which clones `avatar-qemu` and tries to build it. When using the `./avatar2/targets/build_qemu.sh`, everything works. I tried to figure out the difference, but did not find any... I will stick with the compiled binary from the `build_qemu.sh` script. Thanks!

mariusmue, 2019-04-12:
Unfortunately, I can't reproduce the issue, even in a freshly setup VM:
```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic
```

Would you mind sharing your version of gcc with us? 

Anyhow, I tried to added a fix with https://github.com/avatartwo/avatar-qemu/commit/59cfd6f5df28b0e805cc9561f5ba5f784772a3fd - could you retry to run the installer and confirm whether the problem still exists?

phros, 2019-04-15:
I deleted the ~/.avatar2 folder and reran the installer. Installing avatar-qemu works now! Thanks. 

I had an issue building avatar-panda.. I will issue a new issue for this.

closed 2019-04-15 by phros