avatartwo/avatar2 #30: Using already running qemu

phros, 2019-04-12:
Hi Guys,

I try to learn avatar2 in order to analyse IoT software.
My setup, which I set up manually:
* ARM qemu running Raspbian
* chroot inside the qemu with all required files from IoT device
* webserver-binary running 
* gdbserver attached to running webserver

I managed to connect to the gdbserver with avatar2, but I wonder how this is the correct way to go. 
Is it possible to create a QemuTarget for the already running Qemu process? 
Or shall I do something completely different?

Every kind of advice or feedback is welcome :)

Cheers

mariusmue, 2019-04-12:
Hi there!
For now, QemuTarget would be responsible for spawning the QemuInstance.

If you want to attach to another Qemu instance, you could advice Qemu to spawn a GDBServer as well, and attach to it using the GDBTarget! Just add:
`-gdb tcp::127.0.0.1:[PORT]` to your command line arguments for Qemu. Additionally, if you want to halt the CPU on startup, add also the `-S` flag. Note that this way, you won't be able to access features of avatar-qemu like remote memory.

If you need access to Qemus QMP, you could set the Monitor protocol of your GDBTarget to qmp.

This may work like this (assuming the port for qmp is 9999):
```
qmp = QMPProtocol(9999, origin=self) 
gdbtargetInstance.protocols.monitor= qmp
qmp.connect()
```
This would require your Qemu instance started with the qmp server:
`-qmp tcp:127.0.0.1:9999,server,nowait`

Note that this is not tested and just written from my memory - normally we use the QemuTarget to drive QEMU, and we haven't implemented a `RemoteQemuTarget` yet - but this could be totally done the way described above. :)

phros, 2019-04-12:
Thanks for the feedback, I will give it a try!

mariusmue, 2021-02-09:
As there where no further comments, I'm closing the issue for now.

closed 2021-02-09 by mariusmue