avatartwo/avatar2 #31: Building avatar-panda with install script: python issue

phros, 2019-04-15:
Hi, cloned avatar2 and installed it in a python3 virtual env. Then I use the installer (`python -m avatar2.installer`) in order to build avatar-panda.

I got the following error which makes sense because I use python3:
```
ERROR: Cannot use 'python', Python 2.6 or later is required.
       Note that Python 3 or later is not yet supported.
       Use --python=/path/to/python to specify a supported Python.
```
I wonder how to proceed. Shall I use python2 for avatar in general or only for the installer? I could also try to clone/build avatar-panda without the installer. Or i

mariusmue, 2019-04-15:
This requirement is for qemu/panda, not for avatar2, as their build-system requires python2.
Hence, you can add it to the configure options in the installer. This step isn't automated yet, as it would need to resolve the location of the python2.

Additionally, there is the hope that this problem fixes itself from alone, as py2 hits its end of life soon. :)
Nevertheless, we would happily accept a pull request for this.

Screenshot:
[image]

phros, 2019-04-15:
Thanks, didn't thought to specify the path at this point.. my mistake.

closed 2019-04-15 by phros