avatartwo/avatar2 #35: Unsupported value for dictifying executable for target OpenOCDTarget0

mksavic, 2019-05-13:
### Description
Building and installing Avatar2 out of the box and then running the `nucleo_l152re` example in [avatartwo/avatar2-examples](https://github.com/avatartwo/avatar2-examples) does not work on Ubuntu 16.04.

### Reproduce
```
$ sudo apt-get install python-pip python-setuptools python-dev cmake
$ git clone https://github.com/avatartwo/avatar2.git
$ cd avatar2/
$ sudo python setup.py install
$ cd targets/
$ build_qemu.sh
$ build_panda.sh
```
```
$ git clone https://github.com/avatartwo/avatar2-examples.git
$ cd avatar2-examples/nucleo_l152re/
$ python nucleo_state_transfer.py
```

### Error
```
$ python nucleo_state_transfer.py 
Traceback (most recent call last):
  File "nucleo_state_transfer.py", line 88, in <module>
    main()
  File "nucleo_state_transfer.py", line 38, in main
    avatar.init_targets()
  File "build/bdist.linux-x86_64/egg/avatar2/avatar2.py", line 207, in init_targets
  File "build/bdist.linux-x86_64/egg/avatar2/targets/qemu_target.py", line 166, in init
  File "build/bdist.linux-x86_64/egg/avatar2/targets/qemu_target.py", line 115, in generate_qemu_config
  File "build/bdist.linux-x86_64/egg/avatar2/avatar2.py", line 134, in generate_config
  File "build/bdist.linux-x86_64/egg/avatar2/targets/target.py", line 239, in dictify
Exception: Unsupported value for dictifying executable for target OpenOCDTarget0
```

### Proposed Solution
Updating `avatar2/avatar2/targets/target.py:228` from `expected_types = (str, bool, int, list)` to `expected_types = (str, bool, int, list, unicode)` fixes the error, however I am unsure if the change maintains correctness.

mariusmue, 2019-05-13:
Hi!
Thanks a lot for reporting!
From what I see, the proposed fix will only work for python2, as python3 does not use a `unicode` type.
Can you check if the issue exists also for avatar running with the python3 interpreter?
If this issue is only existent for python2, a patch would need to check what the current interpreter is and append `unicode` to the expected types list if required. If the issue also existent for python3, feel free to post the according error messages. 

I'd be happy about a PR for this, otherwise I dig into this within the next days.

mariusmue, 2019-05-14:
Fixed with d188e1bc9f4785722d6d7dc95b80dee528e6f983.
Indeed, this bug only lived in python2 - please note that the recommended way to run avatar² is python3.

In any case, thanks a lot for reporting, the reproducible test case, and the suggested fix.

closed 2019-05-14 by mariusmue