avatartwo/avatar2 #40: setup.py fails on debian 9

mwenzl, 2019-11-21:
Hi,

setup.py ('python setup.py install --user --prefix=')  fails on debian 9 with the message:

```log
[ 95%] Building CXX object llvm/keystone/CMakeFiles/keystone.dir/__/lib/Target/X86/MCTargetDesc/X86AsmBackend.cpp.o
[ 95%] Building CXX object llvm/keystone/CMakeFiles/keystone.dir/__/lib/Target/X86/MCTargetDesc/X86ELFObjectWriter.cpp.o
[ 96%] Building CXX object llvm/keystone/CMakeFiles/keystone.dir/__/lib/Target/X86/MCTargetDesc/X86MCAsmInfo.cpp.o
[ 97%] Building CXX object llvm/keystone/CMakeFiles/keystone.dir/__/lib/Target/X86/MCTargetDesc/X86MCCodeEmitter.cpp.o
[ 97%] Building CXX object llvm/keystone/CMakeFiles/keystone.dir/__/lib/Target/X86/MCTargetDesc/X86MCTargetDesc.cpp.o
[ 98%] Building CXX object llvm/keystone/CMakeFiles/keystone.dir/__/lib/Target/X86/TargetInfo/X86TargetInfo.cpp.o
[ 99%] Building CXX object llvm/keystone/CMakeFiles/keystone.dir/ks.cpp.o
[100%] Linking CXX shared library ../lib/libkeystone.so
[100%] Built target keystone
error: Setup script exited with error: SandboxViolation: mkdir('/usr/lib/python2.7/dist-packages/keystone', 511) {}
```

Please find a patch solving this issue below.

Best,
mwenzl

[setup.py.patch.txt](https://github.com/avatartwo/avatar2/files/3874350/setup.py.patch.txt)

```diff
diff --git a/setup.py b/setup.py
index 29be93b..0a27039 100644
--- a/setup.py
+++ b/setup.py
@@ -1,5 +1,19 @@
 from setuptools import setup
 from sys import version_info
+from setuptools.command.install import install
+
+class CustomInstallCommand(install):
+    def run(self):
+        try:
+            from setuptools.sandbox import DirectorySandbox
+            def violation(operation, *args, **_):
+                print "SandboxViolation: %s" % (args,)
+            
+            DirectorySandbox._violation = violation
+        except ImportError:
+            pass
+        
+        return install.run(self)
 
 
 setup(
@@ -31,5 +45,6 @@ setup(
     url='https://github.com/avatartwo/avatar2',
     description='A Dynamic Multi-Target Orchestration Framework',
     maintainer='Marius Muench',
-    maintainer_email='marius.muench@eurecom.fr'
+    maintainer_email='marius.muench@eurecom.fr',
+    cmdclass={'install': CustomInstallCommand}
 )
```

mariusmue, 2019-11-21:
Thanks for reporting! This seems to be yet another problem related to the keystone project (and not avatar2).
In any case, I don't think it is wise to just ignore Sandbox errors away and I would prefer to not merge an according patch.

As keystone is only required for the assembler plugin, I think it is about time to decouple the setup of avatar2 from capstone/keystone. I can look into it once I have spare cycles.

mariusmue, 2020-07-14:
The latest version of keystone on [PyPi](https://pypi.org/project/keystone-engine/0.9.2/) (v0.9.2) seems to fix the keystone-related installation errors.
Hence, I'm closing this issue for now - if there still persist problem, please add a comment here.

closed 2020-07-14 by mariusmue