avatartwo/avatar2 #53: QEMU: Terminated via GDBstub

Hvnt3r, 2020-05-16:
I get an error when I use the command：
`qemu-arm -singlestep -g 1235 ./vuln3_debug`

Code：

```
import os
import subprocess

from avatar2 import *

filename = 'vuln3_debug'
GDB_PORT = 1235   

shellcode = (b'\x68\x72\x6c\x64\x21\x48\xb8\x48\x65\x6c\x6c\x6f\x20\x57\x6f\x50'
             b'\x48\x89\xef\x48\x89\xe6\x6a\x0c\x5a\x6a\x01\x58\x0f\x05')
 

# Create the avatar instance and specify the architecture for this analysis
avatar = Avatar(arch=archs.arm.ARM)

# Create the endpoint: a gdbserver connected to our tiny ELF file
gdbserver = subprocess.Popen('qemu-arm -g %d vuln3_debug' % GDB_PORT, shell=True)

# Create the corresponding target, using the GDBTarget backend
target = avatar.add_target(GDBTarget, gdb_port=GDB_PORT)

target = avatar.add_target(QemuTarget, 
    gdb_port=GDB_PORT,executable = "/usr/bin/qemu-arm",
    gdb_executable = "gdb-multiarch")

# Initialize the target. 
# This usually connects the target to the endpoint
target.init()
bkpt = qemu.set_breakpoint(0x103fe)
# Now it is possible to interact with the target.
# For example, we can insert our shellcode at the current point of execution
target.write_memory(target.read_register('pc'), len(shellcode),
                    shellcode, raw=True)

# We can now resume the execution in our target
# You should see hello world printed on your screen! :)
target.cont()

# Clean up!
#os.remove(filename)
avatar.shutdown()
```

error：

```
➜  my_code python3 x86_fuzz.py
Traceback (most recent call last):
  File "x86_fuzz.py", line 29, in <module>
    target.init()
  File "/usr/local/lib/python3.6/dist-packages/avatar2/watchmen.py", line 78, in watchtrigger
    ret = func(self, *args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/avatar2/targets/qemu_target.py", line 206, in init
    if gdb.remote_connect(port=self.gdb_port) and qmp.connect():
  File "/usr/local/lib/python3.6/dist-packages/avatar2/protocols/qmp.py", line 31, in connect
    self._telnet = telnetlib.Telnet('127.0.0.1', self.port)
  File "/usr/lib/python3.6/telnetlib.py", line 218, in __init__
    self.open(host, port, timeout)
  File "/usr/lib/python3.6/telnetlib.py", line 234, in open
    self.sock = socket.create_connection((host, port), timeout)
  File "/usr/lib/python3.6/socket.py", line 724, in create_connection
    raise err
  File "/usr/lib/python3.6/socket.py", line 713, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
qemu-arm: QEMU: Terminated via GDBstub
```

**Can someone tell me how to solve this error, thank you very much**

mariusmue, 2020-05-17:
You are using qemu-arm (usermode) outside the context of avatar, hence you don't need the avatar QemuTarget, but just the GDBTarget.
If you ever want to use full-system-qemu with avatar2, be aware that you will need avatar-qemu, and not the one of your distribution.

In any case, User-mode qemu is not supported as target in avatar2 yet, and hence, must be interfaced via the gdb target. Following this, you have to remove this lines:
```
target = avatar.add_target(QemuTarget, 
    gdb_port=GDB_PORT,executable = "/usr/bin/qemu-arm",
    gdb_executable = "gdb-multiarch")
```

Furthermore, the gdbserver you are spawning is around the qemu-process, not the binary you are running - which is most likely not what you want.
You most likely want to run your qemu-user with the -g flag, to specify a port on which a gdbserver *to* the emulated binary will be opened.

Hvnt3r, 2020-05-17:
Thanks for your answer, it really helped me a lot, good luck！

closed 2020-05-18 by mariusmue