avatartwo/avatar2 #58: where is  the angr target?

jinjing750629, 2020-09-24:
Dear Sir

I saw there is a angr target inside avatar2's publication. But I can't find it anywhere in the source code. How can I connect angr to get symbolic execution from avatar2? 

Thanks

mariusmue, 2020-09-24:
We currently don't have an active maintainer for the angr-target.
Hence, it is out of date/based off an old version of angr. You can find the code in this branch: https://github.com/avatartwo/avatar2/tree/dev/angr_target

When it comes to the paper, the full examples, including code, are at https://github.com/avatartwo/bar18_avatar2 - which uses avatar2 at the code tag https://github.com/avatartwo/avatar2/tree/bar18_avatar2.

AdamVanScyoc, 2021-02-28:
Isn't that what is? https://github.com/angr/angr-targets

mariusmue, 2021-03-01:
No, these are the targets of angr, which reimplement the avatar2 target API for interoperability. So, it's the way angr can talk with different software, e.g, avatar2 - and not the other way around.

AlexCasual, 2022-06-06:
Hi, what about adopt old version of angr-target for new version of angr?