avatartwo/avatar2 #59: Could we only use one OpenOCD target to get the mmio in avatar2?

DarknessChen, 2020-09-25:
I want to get the mmio recordings while the board is running (instead of doing the rehosting). Is it possible to get the mmio (read/write) traffic by using avatar2? I modify some codes from pretender (e.g. set the memory map but setting the `forwarded` argument of `add_memory_range` to `false`, however, there is no recordings in the output). Thanks!

mariusmue, 2020-09-28:
Heya,
The recording/logging of MMIO is based on the fact that MMIO accesses are made *via* avatar2.
Remote Memory Requests are emitted by one target and send to avatar2, which then logs them and forwards it to the according target.

Hence, avatar2 can currently not log MMIO accesses if they just happen local on the device.

I think the best way to get those accesses is collecting ETM traces on the board itself, but it requires support from the hardware side and an according debugger.

DarknessChen, 2020-09-29:
This solves my problem. Thanks a lot @mariusmue ! I think Avatar2 is a really good community:)

closed 2020-09-29 by mariusmue