avatartwo/avatar2 #60: Rehosting the Raspbian

DarknessChen, 2020-10-03:
Hi,

May I ask whether avatar2 is able to rehost the OS-based firmware (e.g. Raspbian or Openwrt) or not? I notice that it is able to deal with armv7-M architecture (for microcontroller), but how about armv7-A/armv8-A (for application e.g. raspberry pi)? 

If it is possible, how to set up the memory map? For the MCU, it is not difficult to figure out the base of rom (likely be `0x08000000`), ram (`0x20000000`) and mmio (`0x40000000`). I know there is a difference between bare-metal program and OS-based kernel. However, when I search the memory mapping of raspberry pi, I got [this picture](https://www.raspberrypi.org/forums/viewtopic.php?t=262747) and it is still confused for me on how to set up the mem map in avatar2.

Thankful for any answer/idea!

rawsample, 2020-10-04:
Hello @DarknessChen,

I'm afraid that avatar does not support the re-hosting of a full OS-based firmware for now.

On cortex-a, the application does not access the physical memory by itself like it would do in cortex-m. Instead the OS with the help of the MMU create a virtual address space for the application and map it on the physical one. This mapping can change over time. So for rehosting, you have to made a choice on which side of the memory abstraction you put yourself.
To keep things harder, you also have the _Translation Lookaside Buffer_ (TLB) in the MMU that act as a cache for physical addresses which were recently translated.
If you would like to start digging more on how MMU works with ARM, I invite you to follow this link:
https://developer.arm.com/architectures/learn-the-architecture/memory-management/single-page

Bests

mariusmue, 2020-10-05:
Yes, it's not supported out of the box and needs likely some changes, especially if you want to use memory forwarding to/from a real device. However (!) you can use avatar in any case for orchestrating a QEMU/PANDA-instance running raspbian/openwrt; the API of read/write/breakpoints/cont/stop/etc also works in this cases.
Only knowledge about the memory layout is not supported, as for now, we only deal with flat & monolithic memory spaces.

DarknessChen, 2020-10-06:
Thanks a lot @rawsample @mariusmue !

closed 2021-02-09 by mariusmue