avatartwo/avatar2 #61: Dependency problem pygdbmi

icemonster, 2020-10-18:
Avatar2 breaks with the new version of pygdbmi (0.10.0.0).
Installing this on ubuntu >= 19 seems to yield an error when running the "Hello world" example in the tests folder.

After installing a previous version of pygdbmi, everything runs smoothly
`python3 -m pip pygdbmi==0.9.0.2`

mariusmue, 2020-10-19:
Which version of Python, and which version avatar2 were you using when encountering this problem?

Did you test if the problem persists with git-master of avatar2?
Background is, we recently merged #32 which should fix the issue - but didn't update the version on PyPi yet.

mariusmue, 2021-02-09:
Can not reproduce with current master version of avatar2, and no further comments where given here. I'll close this for now, if the problem persist, please let us know!

closed 2021-02-09 by mariusmue