avatartwo/avatar2 #66: GDBProtocol was unable to connect

sh4m2hwz, 2021-01-12:
>>> avatar_gdb = AvatarGDBConcreteTarget(avatar2.archs.x86.X86,"127.0.0.1", 1234)
CRITICAL | 2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol | GDBProtocol was unable to connect to remote target
2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/angr_targets/targets/avatar_gdb.py", line 23, in __init__
    self.avatar.init_targets()
  File "/usr/local/lib/python3.9/dist-packages/avatar2/avatar2.py", line 222, in init_targets
    t[1].init()
  File "/usr/local/lib/python3.9/dist-packages/avatar2/targets/gdb_target.py", line 53, in init
    if gdb.remote_connect(ip=self.gdb_ip, port=self.gdb_port):
  File "/usr/local/lib/python3.9/dist-packages/avatar2/protocols/gdb.py", line 379, in remote_connect
    raise Exception("GDBProtocol was unable to connect")
Exception: GDBProtocol was unable to connect
 
 
-----------------------------------------------------------------------------
 
Attached; pid = 18032
Listening on port 1234
Remote debugging from host 127.0.0.1

mariusmue, 2021-01-12:
We don't have AvatarGDBConcrete Target in our repository. Perhaps you are using code from Symbion/Angr?

In any case, from what I see, there's a mismatch and args you specify and kwarg order expected by avatar2's gdbtarget: https://github.com/avatartwo/avatar2/blob/master/avatar2/targets/gdb_target.py#L9

But then, I don't know if the ConcreteTarget you are using uses the same API.

Cheers,
Marius

sh4m2hwz, 2021-01-12:
@mariusmue hello, yes, I am using Angr Symbion, but the avatar gdb is used as an interface, if I am not mistaken. If you look at the constructor prototype, then the arguments are all correct:
https://github.com/angr/angr-targets/blob/62392168d45c3cdc8021cf1ec9146eaf812bb325/angr_targets/targets/avatar_gdb.py#L18

mariusmue, 2021-01-12:
Are you using the git version of avatar2, or the pypi one? I recall a significant API change in pygdbmi, which is integrated here in the git, but not on pypi.
Other than that, if you try with the actual avatar gdb-target, you can set `gdb_verbose_mi` to true and should receive additional debugging output.
https://github.com/avatartwo/avatar2/blob/master/avatar2/targets/gdb_target.py#L14

Taardisaa, 2021-11-07:
hello, have you 

> 
> 
> > > > avatar_gdb = AvatarGDBConcreteTarget(avatar2.archs.x86.X86,"127.0.0.1", 1234)
> > > > CRITICAL | 2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol | GDBProtocol was unable to connect to remote target
> > > > 2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
> > > > 2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
> > > > 2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
> > > > 2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
> > > > 2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
> > > > 2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
> > > > 2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
> > > > 2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
> > > > 2021-01-12 07:11:52,893 | avatar.targets.GDBTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
> > > > Traceback (most recent call last):
> > > > File "", line 1, in 
> > > > File "/usr/local/lib/python3.9/dist-packages/angr_targets/targets/avatar_gdb.py", line 23, in **init**
> > > > self.avatar.init_targets()
> > > > File "/usr/local/lib/python3.9/dist-packages/avatar2/avatar2.py", line 222, in init_targets
> > > > t[1].init()
> > > > File "/usr/local/lib/python3.9/dist-packages/avatar2/targets/gdb_target.py", line 53, in init
> > > > if gdb.remote_connect(ip=self.gdb_ip, port=self.gdb_port):
> > > > File "/usr/local/lib/python3.9/dist-packages/avatar2/protocols/gdb.py", line 379, in remote_connect
> > > > raise Exception("GDBProtocol was unable to connect")
> > > > Exception: GDBProtocol was unable to connect
> 
> Attached; pid = 18032 Listening on port 1234 Remote debugging from host 127.0.0.1

hello, have you solved this problem? My AvatarGDBConcreteTarget is also unable to connect to the gdbserver. Thanks.

TheSilentDawn, 2022-01-06:
I also have this problem. I tried to follow the solution https://github.com/avatartwo/avatar2/issues/61. But, pip gave a warning "ERROR: avatar2 1.4.6 has requirement pygdbmi>=0.10.0.0, but you'll have pygdbmi 0.9.0.2 which is incompatible.". Also, while running demo, another error "AttributeError: 'GDBProtocol' object has no attribute '_communicator'" is shown.

sh4m2hwz, 2022-01-06:
> 

No solved

TheSilentDawn, 2022-01-07:
@sh4m2hwz @Taardisaa I have solved my problem. But it's not the problem of the avatar. For me, it's the QEMU configuration problem. You could insert instruction `self.log.info(cmd_line)` here https://github.com/avatartwo/avatar2/blob/main/avatar2/targets/qemu_target.py#L108 and install avatar module to python from source code. Only avatar2 module installation from source code is necessary. Other parts could install by `python -m avatar2.installer`
You will get the QEMU startup command line from the log output. Copy and run this command line in the terminal directly, you will get an output. If the error occurs, parse it here, or you may find the problem by yourself.

b1ack0wl, 2023-07-18:
Cross posting just in case this helps out OP: https://github.com/avatartwo/avatar2/issues/105#issuecomment-1639119902