avatartwo/avatar2 #67: Missing 'cpu_mips_init' function for avatar-qemu MIPS

kilate, 2021-01-24:
Getting an error when building avatar-qemu  for a MIPS target:

```
avatar2/targets/src/avatar-qemu/hw/avatar/configurable_machine.c: In function ‘create_cpu’:
avatar2/targets/src/avatar-qemu/hw/avatar/configurable_machine.c:513:12: error: implicit declaration of function ‘cpu_mips_init’; did you mean ‘cpu_mmu_index’? [-Werror=implicit-function-declaration]
     cpuu = cpu_mips_init(cpu_model);
            ^~~~~~~~~~~~~
            cpu_mmu_index
avatar2/targets/src/avatar-qemu/hw/avatar/configurable_machine.c:513:12: error: nested extern declaration of ‘cpu_mips_init’ [-Werror=nested-externs]
avatar2/targets/src/avatar-qemu/hw/avatar/configurable_machine.c:513:10: error: assignment makes pointer from integer without a cast [-Werror=int-conversion]
     cpuu = cpu_mips_init(cpu_model);
```

Configured with:
`../../src/avatar-qemu/configure  --disable-sdl --target-list=mipsel-softmmu,mipsel-linux-user
`
Could not find cpu_mips_init definition for avatar-qemu (there is a prototype in cpu.h though). Avatar-panda has the definition on tranlate.c but avatar-qemu does not.

If I include:
```
MIPSCPU *cpu_mips_init(const char *cpu_model);
MIPSCPU *cpu_mips_init(const char *cpu_model)
{

    return NULL;
}
```

Inside _configurable_machine.c_ it compiles with no errors.

Using Ubuntu 18.04.
Thanks

mariusmue, 2021-01-24:
Hi!

The MIPS target of avatar is unfortunately not actively maintained. That changes to the configurable machine were created by the PANDA devs. I'm happy to merge an according PR. Can you create it? Otherwise, @rawsample, do you want to look into it?

Best,
Marius

kilate, 2021-01-24:
Hi,
sorry to hear that.
No, I have no skills for that (not a developer/programmer).

Thanks anyway.

kilate, 2021-01-24:
Well, I managed to get it compiled copying the functions 'cpu_mips_init' and 'cpu_mips_init' from avatar2-panda's (target/mips/translate.c) into avatar2-qemu and doing some other fixes (diff below. Sorry, I'm learning this git stuff yet, so no PR). 
So, qemu itself seems to be ok, but now I really understand what you meant by 'not actively maintained' regarding MIPS. Python module(s) is/are missing.  For me is a game over for now.

Hope you get it finished in a near future.  It's a great concept and I learnt few thing messing with it. Thanks for sharing.

```
$ git diff
diff --git a/dtc b/dtc
--- a/dtc
+++ b/dtc
@@ -1 +1 @@
-Subproject commit 85e5d839847af54efab170f2b1331b2a6421e647
+Subproject commit 85e5d839847af54efab170f2b1331b2a6421e647-dirty
diff --git a/target/mips/cpu.h b/target/mips/cpu.h
index 7cf7f5239f..13966341c1 100644
--- a/target/mips/cpu.h
+++ b/target/mips/cpu.h
@@ -1164,7 +1164,7 @@ struct MIPSCPU {
 
 
 void mips_cpu_list(void);
-
+MIPSCPU *cpu_mips_init(const char *cpu_model);
 #define cpu_signal_handler cpu_mips_signal_handler
 #define cpu_list mips_cpu_list
 
diff --git a/target/mips/translate.c b/target/mips/translate.c
index 9fad58ea2c..b8522e2f03 100644
--- a/target/mips/translate.c
+++ b/target/mips/translate.c
@@ -31324,6 +31324,46 @@ void mips_tcg_init(void)
 
 #include "translate_init.inc.c"
 
+
+static const mips_def_t *cpu_mips_find_by_name (const char *name);
+
+static const mips_def_t *cpu_mips_find_by_name (const char *name)
+{
+    int i;
+
+    for (i = 0; i < ARRAY_SIZE(mips_defs); i++) {
+        if (strcasecmp(name, mips_defs[i].name) == 0) {
+            return &mips_defs[i];
+        }
+    }
+    return NULL;
+}
+MIPSCPU *cpu_mips_init(const char *cpu_model)
+{
+    MIPSCPU *cpu;
+    CPUMIPSState *env;
+    const mips_def_t *def;
+
+    def = cpu_mips_find_by_name(cpu_model);
+    if (!def)
+        return NULL;
+    cpu = MIPS_CPU(object_new(TYPE_MIPS_CPU));
+    env = &cpu->env;
+    env->cpu_model = def;
+    env->exception_base = (int32_t)0xBFC00000;
+
+#ifndef CONFIG_USER_ONLY
+    mmu_init(env, def);
+#endif
+    fpu_init(env, def);
+    mvp_init(env, def);
+
+      object_property_set_bool(OBJECT(cpu), "realized", true, NULL);
+
+    return cpu;
+}
```

mariusmue, 2021-01-25:
Hey, great to hear you got it compiling, and even more happy that you learned new things along the way!
On the Python side of things, all it needs is additional addition in the archs submodule. For instance, compare here against ARM:
https://github.com/avatartwo/avatar2/blob/master/avatar2/archs/arm.py

After that, you should be ready to go.

rawsample, 2021-01-25:
Hi,

Thanks @kilate, I transformed it into a commit.
https://github.com/avatartwo/avatar-qemu/commit/9abf34ceda9f05e81eed46d6fbf07657195c9dd5

mariusmue, 2021-02-09:
Merged and extended with 90547c193d888614686d10e9e1fcf56b4ebf228d! 
Thanks @rawsample!
@kilate, you now also have the Python side in here and could try your luck.

closed 2021-02-09 by mariusmue

kilate, 2021-02-10:
Hi,
thanks a lot for this. I will try it for sure.