avatartwo/avatar2 #78: Feature request: Add ability to generate core dump in GDB

AdamVanScyoc, 2021-02-28:
Hi, thanks for avatar, it's highly interesting.

I was wondering if there would be interest in a feature that lets you generate a core dump in GDB.

My use case would look like this: Say I'm using angr's exploration technique "Symbion" (https://angr.io/blog/angr_symbion/) to dynamically analyze a binary. After angr has stepped through some of the code, at a certain point I might like to create a GDB core dump of a certain state that angr has produced, which would let me do further manual dynamic analysis in GDB.

If this seems like an OK idea, I'd take a crack at implementing it and then submit a PR.

Thanks again

mariusmue, 2021-03-01:
Hi @AdamVanScyoc!

Sounds like a great feature to be implemented as plugin! I'd be happy to merge an according PR. I think the closest plugin is the [gdb-memory-map-loader](https://github.com/avatartwo/avatar2/blob/main/avatar2/plugins/gdb_memory_map_loader.py).

Unfortunately, we don't have good documentation on how to write a plugin, but I hope the link above gives you a good overview over the programming model. Alternatively, if you run into problems, I'm happy to answer your questions.

Thank you for your help!

AdamVanScyoc, 2021-03-11:
Would there be any interest in an extension to this that allows a user to restore the state of a program from a core file for a GDB target?

mariusmue, 2021-03-12:
Sure, we are generally happy about new features as plugins. If you feel like adding tests to our CI for your plugin, we would even make sure that it does not break in future versions of avatar2.

If you need help on how to create those tests, I'm happy to give pointers.