avatartwo/avatar2 #101: PyPanda: avoid duplicate monitor

grant-h, 2022-02-16:
PyPanda's Panda() top-level class by default creates an interactive  monitor like `-monitor unix:/tmp/pypanda_mbwril7yr,server,nowait`. I dont believe monitor is not the same as QMP, but a text based protocol

There are couple of ideas:

* Always disable PyPanda's monitor by passing `raw_monitor=True` (https://github.com/panda-re/panda/blob/716b782f3f604d990f361cbab3186b4e0c73540f/panda/python/core/pandare/panda.py#L59)
* Provide an option to PyPanda target creation to disable this by default, but still enable it if needed

rawsample, 2022-02-17:
Hi!

Disabling PyPanda's monitor should be currently possible via `additional_args` in the [QemuTarget](https://github.com/avatartwo/avatar2/blob/25de83f961b9474ee418d8844bf055f2428527cb/avatar2/targets/qemu_target.py#L29) because of  `assemble_cmd_line()` [here](https://github.com/avatartwo/avatar2/blob/main/avatar2/targets/pypanda_target.py#L46).

But i would be in favor on changing the default behavior via a new option in PyPanda target.

grant-h, 2022-02-17:
I tried that, but doesn't raw_monitor needs to be passed as a kwarg, not an array of strings. Can you show me?