avatartwo/avatar2 #105: Exception: GDBProtocol was unable to connect

bukeyong, 2022-04-19:
When I try to run `python3 test_pyperipheral.py`, I run into the following problem：

2022-04-19 06:33:03,450 | avatar.INFO | Initialized Avatar. Output directory is /tmp/avatar
2022-04-19 06:33:03,456 | avatar.targets.QemuTarget0.INFO | QEMU process running
2022-04-19 06:33:08,531 | avatar.targets.QemuTarget0.GDBProtocol.CRITICAL | GDBProtocol was unable to connect to remote target
Traceback (most recent call last):
  File "test_pyperipheral.py", line 121, in <module>
    setup_func()
  File "/home/desh0ng/.local/lib/python3.8/site-packages/nose/tools/nontrivial.py", line 97, in newfunc
    result = func(*arg, **kw)
  File "test_pyperipheral.py", line 47, in setup_func
    avatar.init_targets()
  File "/usr/local/lib/python3.8/dist-packages/avatar2-1.4.7-py3.8.egg/avatar2/avatar2.py", line 233, in init_targets
    t[1].init()
  File "/usr/local/lib/python3.8/dist-packages/avatar2-1.4.7-py3.8.egg/avatar2/watchmen.py", line 78, in watchtrigger
    ret = func(self, *args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/avatar2-1.4.7-py3.8.egg/avatar2/targets/qemu_target.py", line 274, in init
    self._connect_protocols()
  File "/usr/local/lib/python3.8/dist-packages/avatar2-1.4.7-py3.8.egg/avatar2/targets/qemu_target.py", line 314, in _connect_protocols
    connect_success = connect_success and gdb.remote_connect(port=self.gdb_port)
  File "/usr/local/lib/python3.8/dist-packages/avatar2-1.4.7-py3.8.egg/avatar2/protocols/gdb.py", line 399, in remote_connect
    return self._remote_connect_common('%s:%d' % (ip, int(port)))
  File "/usr/local/lib/python3.8/dist-packages/avatar2-1.4.7-py3.8.egg/avatar2/protocols/gdb.py", line 384, in _remote_connect_common
    raise Exception("GDBProtocol was unable to connect")
Exception: GDBProtocol was unable to connect

mariusmue, 2022-04-19:
Heya. The tests are sensitive to the directory they are run from - all of them are expected to be run from the top-level directory, so it should be: `python3 tests/test_gdbprotocol.py`

That being said, on my local machine, I run into some problems regarding the unix socket version of gdb protocol, but I believe that is currently local to my machine.

bukeyong, 2022-04-19:
thank you very much! I haven't noticed the directory problem.

bukeyong, 2022-04-20:
It seems that the reason for this error is not just the path problem. I found that if the configuration of avatar is wrong, the exception "GDBProtocol was unable to connect" will also be generated.
Hmm~, I'm curious how the entry point in the sample is determined? I tried to unzip the .bin file but it didn't work.

bukeyong, 2022-04-20:
> Heya. The tests are sensitive to the directory they are run from - all of them are expected to be run from the top-level directory, so it should be: `python3 tests/test_gdbprotocol.py`
>
> That being [...]

It seems that the reason for this error is not just the path problem. I found that if the configuration of avatar is wrong, the exception "GDBProtocol was unable to connect" will also be generated.
Hmm~, I'm curious how the entry point in the sample is determined? I tried to unzip the .bin file but it didn't work.

b1ack0wl, 2023-07-18:
I've recently come across this issue and narrowed it down to a hard-coded timeout value in `protocols/gdb.py`. When connecting to a gdbserver instance the client downloads the running binaries from the server which can take longer than 5 seconds. Due to the function `_sync_request()` having a hard-coded timeout value of `5` an exception can be thrown when downloading the binaries from the server.

Command that can take longer than 5 seconds: https://github.com/avatartwo/avatar2/blob/main/avatar2/protocols/gdb.py#L377
Hard-coded argument value: https://github.com/avatartwo/avatar2/blob/main/avatar2/protocols/gdb.py#L304C9-L304C22

I changed my `timeout` value to `30` and it worked perfectly. I hope this helps :crossed_fingers: