avatartwo/avatar2 #107: Is it possible to dynamically analyse the firmware of a ESP32 just with the default avatar2 tools?

peterrakolcza, 2022-05-17:
Hey!

After tinkering with avatar2, I still cannot really debug a ESP32.:(

I have a ESP32 with a very simple firmware flashed to it, I am using a ES-Prog as a JTAG adapter. Following the demonstration video from ACM WiSec 2021, I copied the followint code:
```
from avatar2 import Avatar
from avatar2.archs import #architeture
from avatar2.targets import OpenOCDTarget

avatar = Avatar(arch= , output_directory='/tmp/avatar2')

#Specify target
esp = avatar.add_target(OpenOCDTarget, openocd_script='')

#Add memory
#Skip

avatar.init_targets()

#esp.set_breakpoint()
esp.cont()
esp.wait()

import IPython; IPython.embed()
```

However, I do not know what architecture should I define fot the ESP32 and what OpenOCD target script should I use. Any help is appreciated!

aurelf, 2022-05-17:
Hi, I depends on the ESP model you are using, some are using xtensa others (recent ones) Risc-V (and some have several cores some xtensa some Risc-V). Which model is your ESP32 exactly ?
 
Either way I don't think we have any specific support for xtensa or Risc-V for now, this would significantly limit what's possible to do. A support for those architectures would be very welcome :)

peterrakolcza, 2022-05-17:
Hi! Thanks for your input! I have a NodeMCU-ESP-32S. How hard would it be to develop support for this architecture using existing code?

peterrakolcza, 2022-05-17:
What is a cheap microcontroller which is supported by the avatar2 platform and can be used to learn about dynamic firmware analysis?

mariusmue, 2022-05-17:
Hi, and thanks for your interest in avatar2!

As @aurelf mentioned, we currently don't have an ESP32 target. A lot of avatar2's original implementation and examples where done on top of STM32'2 Cortex-M3 MCUs, specifically on the NUCLEO-L152RE.
While these are currently unavailable in most stores, similar usually do work with avatar2 as well.

closed 2022-05-17 by peterrakolcza