avatartwo/avatar2 #113: RecursionError: maximum recursion depth exceeded in comparison

mehdibaghery, 2022-11-14:
when I want to connect QemuTarget I face to this error

rawsample, 2022-11-14:
Hello,

To help you, could you provide more context on the issue? Could you share the full backtrace? and the code so I can try to reproduce it.

mehdibaghery, 2023-01-28:
In [2]: avatar.transfer_state(ocd, qemu, synced_ranges=[ram])
---------------------------------------------------------------------------
RecursionError                            Traceback (most recent call last)
Cell In [2], line 1
----> 1 avatar.transfer_state(ocd, qemu, synced_ranges=[ram])
.
.
.

File ~/Desktop/test_venv/venv_avatar/lib/python3.10/site-packages/pygdbmi-0.10.0.2-py3.10.egg/pygdbmi/IoManager.py:236, in IoManager.write(self, mi_cmd_to_write, timeout_sec, raise_error_on_timeout, read_response)
    223 """Write to gdb process. Block while parsing responses from gdb for a maximum of timeout_sec.
    224 
    225 Args:
   (...)
    233     TypeError: if mi_cmd_to_write is not valid
    234 """
    235 # self.verify_valid_gdb_subprocess()
--> 236 if timeout_sec < 0:
    237     logger.warning("timeout_sec was negative, replacing with 0")
    238     timeout_sec = 0

RecursionError: maximum recursion depth exceeded in comparison