avatartwo/avatar2 #121: questions regarding gdbstub, SerialICE and avatar[one]

hanetzer, 2023-07-30:
So, I came across this project while fiddling around with SerialICE for coreboot dev purposes.
The handbook mentions avatar2 can still do a GDBTarget, but as far as I can see there is no
gdbstub firmware under the avatartwo arg; am I to believe I should use the avatar-gdbstub
from the old project?

And on a related note, best as I can tell, the old avatar-gdbstub only supports arm, is this
correct?

Also, is it possible to do SerialICE-style io logging in avatar2?

All in all a very interesting project so far, and assuming my gdbstub question is answered
in the affirmative, it would be interesting to make use of on some of my iot reversing projects.

Regards.

mariusmue, 2023-07-31:
Heya,

Thanks for your interest in the avatar2 framework. The way avatar2's GDBTarget works is that it spawns a local GDB instance on the analysis host and communicates with it via GDB Machine Interface ([GDBMI](https://sourceware.org/gdb/onlinedocs/gdb/GDB_002fMI.html)). 

This local instance then in turn communicates with whatever GDB stub your physical target provides; this could be a stub provided by OpenOCD via JTAG, or something like the avatarone gdb stub which uses GDB's remote serial interface ([RSP](https://sourceware.org/gdb/onlinedocs/gdb/Remote-Protocol.html)).

I'm personally not familiar with SerialICE, but one can configure avatar2 to log peripheral accesses and/or forward IO accesses to python reimplementations of hardware logic. Feel free to check out the [examples](https://github.com/avatartwo/avatar2-examples/tree/master) for more information on that.

Hope these answers help!

hanetzer, 2023-08-01:
ok, so avatarone's gdbstub firmware is relevant to avatar2, good to know.

Serialice is a duo of a patched qemu and an x86 bios, where the 'real' vendor bios is ran in qemu, with io and configuration type stuff being forewarded down to the serialice bios running on the real hardware, all the while all such io and things are logged on the system running serialice's patched qemu. Its meant for reversing vendor bioses to aid in things like porting coreboot to new boards.