avatartwo/avatar2 #122: Failure to check for qemu failure at startup

GregIthaca, 2023-08-10:
In the code here: https://github.com/avatartwo/avatar2/blob/main/avatar2/targets/qemu_target.py#L271
there is no test that QEMU is actually running at the end of the operation. There are various misconfiguration issues (including some discussed elsewhere such as #120) which can cause a quick exist, but the error message that is displayed is
```
Exception: GDBProtocol was unable to connect
```
which does not lead one to understanding where the failure occurred. Only by looking at the error log does one detect the issue.

There isn't a great way around this with subprocess.Popen, but something along these lines is possible if you're willing to wait an extra second for a successful startup:
```
        try:
            rc = self._process.wait(timeout=1)
            if rc is not None:
                self.log.warning("QEMU process terminated prematurely")
                return
        except subprocess.TimeoutExpired:
            pass
```

GregIthaca, 2023-08-10:
As far as I can tell, this is only marginally related to #66 or #105. The same error message occurs regardless of which of a number of different reasons the configuration failed. The problem noted here is fundamentally that the error message doesn't tell you what actually went wrong, if the problem was QEMU failing to start.

rawsample, 2023-08-28:
Hello, thanks for your interest avatar2 :)

The error message is not very clear in the sense that it does not invite to check the stdout/stderr of QEMU. These files are located in the avatar `output_directory`. By default, the redirections are files `QemuTarget0_out.txt` & `QemuTarget0_err.txt` under the folder `/tmp/<something>_avatar/`

Otherwise, I'm not a big fan of using timeout for synchronization: they tend to show different behaviors on different machines as it is mentioned in https://github.com/avatartwo/avatar2/issues/105#issuecomment-1639119902