avatartwo/avatar2 #129: Does avatar2 support Arm64?

Anananwd, 2024-10-10:
I heard that avatar2 supports multi-architecture operation. So I want to use it on the arm64 architecture in my project, but I didn't find the relevant options in the source code to help me switch to the arm64 architecture. Does avatar support the arm64 architecture? If so ,could you please provide some information for me.

Thank you !

aurelf, 2024-10-10:
Hi,  this should be feasible, but not done yet. This was recently discussed in #126

Anananwd, 2024-10-11:
> Hi, this should be feasible, but not done yet. This was recently discussed in #126

Thanks, I have found a avatar2 paper with ARM64 and applied its source code to my project successfully. Here is [EL3XIR](https://github.com/HexHive/EL3XIR)

aurelf, 2024-10-16:
Indeed ! It would be nice to merge this support back here !

mariusmue, 2024-10-16:
Hey all, sorry for not chiming in earlier. I'm aware of EL3XIR and already in contact with the authors - getting this contributed back is not completely unlikely and may happen in the near future. :)

dummys, 2025-04-11:
Hi, do you know when this PR will be accepted ?

aurelf, 2025-04-11:
It would be nice, and if you could test it and report success/failure (with details) that would be a great step towards merging it !